ClubRobotInsat/info-archive #8: Improvement of the communication structs parsing using a specialized library

Terae, 2018-11-20:
This project is based on the **Communication layer** between elecs and infos.

## Actual solution
Actually, a C-file is shared between both Rust and C++ codes. It implements for each module:
* a shared structure
* a **write** function
* a **read** function
* the returned frame-size function

We need to carefully test these functions because of the C usage. It is convenient because of its sharealibity: we can write once these functions and it is used on-the-fly by the two parts.

However, it takes time to write and understand and so it is difficult to write new modules' parser.

## New solution
Instead of reinvent again the wheel, we could use a specialized library to share efficiently data. For exemple, we can use CBOR, [msgpack](https://msgpack.org/) or bincode to make the job.

The only one constraint is to find a good library existing both in C++ and in Rust ***with no-STD support***.

Finally, you need to create your own protocol for each module using this new sharing protocol and make usefull documentation *(i.e. decide how to structure data)*.

Ideally, you would implement the C++ code, make tests and write a simple specification to rusties.

You can find inspiration [in the C directory](https://github.com/ClubRobotInsat/librobot/tree/master/c_src).

gbip, 2018-11-20:
> Ideally, you would implement the C++ code, make tests and write a simple specification to **rustaceans**.

Fixed. :smile:

Terae, 2018-11-25:
With @gbip , we used directly the JSON format to share data and that looks promising. if we are not limited be the weight of frames this solution will be the best to debug with [Wireshark](https://www.wireshark.org/) :smiley: 

I implemented the first roll in the commit 29842409bbc4aa2e6cf275774c7cb677bbdb0b17, it might need improvement and we need to make documentation about the data normalization.

gbip, 2018-11-25:
Nice, I am working on improving [`serde_json_core`](https://github.com/japaric/serde-json-core/pull/7) and I already made a PR to improve things along the road.

The last problems with this crates is the fact that Rust enums are not yet supported...

Terae, 2018-11-25:
Ok, the `serde` library is nevertheless functional or it's going to ask a lot of work before to include it?

Because we can always use the msgpack crate if ever

gbip, 2018-11-25:
It is functional but some features are not supported.
We can always code a workaround (do not use Rust enumes for example), but I'd prefer to implement those features as it is a challenging project.

Terae, 2019-01-23:
The final solution is to parse modules in JSON format, and share these frames through ethernet.

You can find samples of the serialization on the [`Servos::generate_list_jsons()`](https://github.com/ClubRobotInsat/info/blob/develop/src/robot/Modules/Servos.cpp#L181) function.

closed 2019-01-23 by Terae