ClubRobotInsat/info-archive #10: Create the environment JSON file for the year 2019.

Terae, 2018-11-24:
All the configuration table is saved on an JSON file: it is easy to work with and we do not need to compile each time the project to make a change effective.

These configurations are saved on the [`commun`](https://github.com/ClubRobotInsat/info/tree/develop/src/commun) directory, and previous tables examples are still in the project. For example, [this is the 2018's json file](https://github.com/ClubRobotInsat/info/blob/develop/src/commun/table_2018.json).

Every new year, we need to create this file [using the Cup rules](https://www.coupederobotique.fr/wp-content/uploads/Eurobot2019_Rules_Cup_OFFICIAL_FR.pdf). This task is very important to make working:
* the simulator
* the known environment of the robot
    * emplacement of unauthorized moves
    * emplacement of each object to work with

To realize this job more conveniently, @BynaryCobweb developed a [python graphical application](https://github.com/ClubRobotInsat/pytable) which exports a the json file. Then, the file needs to be saved in the same directory that previous years' configurations.

Remember, after each modification, you should rerun cmake *(in CLion: `Tools -> CMake -> Reload CMake Project`)* to make modifications effective, because the configuration is copied by [this CMakeLists.txt](https://github.com/ClubRobotInsat/info/blob/develop/src/commun/CMakeLists.txt) on the C++ code.

The file needs to contain these data:
- [ ] type, angle, size and dimensions - used by both environment and simulator
- [ ] color, dynamic type, mass, enabled flag - for the simulator
- [ ] enabled flag - for the environment; walls are enabled to make the robot able to calculate its trajectory, but cubes are not because they don't influence the navigation

If it might help you, the JSON file is used [here for the environment](https://github.com/ClubRobotInsat/info/blob/develop/src/robot/Strategy/Environment/environment.cpp#L233) and [here for the simulator](https://github.com/ClubRobotInsat/info/blob/develop/src/simulateur/core/WorldIO.cpp#L15).

BynaryCobweb, 2018-12-03:
To visualize the table in 3D, you can use the simulator, with very few work required.

- Checkout the branch develop, get the last version of the code with `git pull` and compile it
- Given that your json file is named `table.json`, call the simulator as follows
```bash
./Simulator --color green --load table.json
```

You should see the simulator display the table as you specified :)

DouarreQ, 2018-12-13:
Hey j'ai push le début de la table da,s la branche dev-tablejson

DouarreQ, 2019-03-14:
J'ai push une autre version de la table