ClubRobotInsat/info-archive #20: Pump control

Terae, 2019-01-22:
This year, we will use pumps to manipulate objects.

In order to control them, you need to build these layers:
* [ ] Choose the action specifications (*ON / OFF*, multiple pumps management from the same electronic board, ...)
* [ ] Specify the JSON API shared with rusties *(shared data, informatic orders, ...)*
* [ ] Implement the [associated module](https://github.com/ClubRobotInsat/info/tree/develop/src/robot/Modules), in particular convenient functions and `private`d `vector<JSON> generate_list_jsons() const` & `void message_processing(const JSON&)` linke[ ]d with the previous point
* [ ] [unit-test](https://github.com/ClubRobotInsat/info/tree/develop/test)ing the new module
* [ ] Develop an [interface](https://github.com/ClubRobotInsat/info/tree/develop/src/robot/Strategy/ModuleInterfacers) able to manage `Pump` and `Servos` modules to make a high-level utility class interfacer. This abstraction layer aims to use 'simple' modules to allow a final user to call functions like `catch-objects()` which manipulate servos and pumps. Actually, this layer is more or less in conception so the intentions for this specific module are blur, we might ponder on this point later when the pump module will be usable.

Ultimately, the objective is to furnish very high-level functionnalities such as `navigate_to(Vector2m)`, `open_right_hold()` or even `catch_right_pucks()` / `release_pucks()` in a graphical-friendly API to build complex strategies.

This  issue would be a formative issue on code abstraction layers using already-existing examples, and the **deadline is the 1st April** *(a working pumps code would be very usefull for the beginning of real-world global tests)*.

Do not hesitate to create new [pull requests](https://github.com/clubrobotinsat/info/pulls) for each step of this issue! :smile:

AbdelSaTd, 2019-02-08:
Hi Terae, you can take a look in the *dev-pump* branch when you will be available. ^^

Terae, 2019-02-12:
I made the review directly on comments of your commit 0d9eddecd681bfed0f805289b8b5e5d238b164c6.

Terae, 2019-03-25:
b5282eeac2a2a16168e445df4266f1a6bbf9a17d integrates the interfacer inside Strategy.

closed 2019-03-25 by Terae